EOA-Blockchain-Labs/ethereumonarm #2: Two /var/log directories, one gets full very fast

handelaar2, 2021-04-01:
Installed the latest image (ethonarm_21.03.00) on my raspberry pi. I noticed that it has created two var/log directories
(/var/log and /var/log.hdd)
This was not the case in the previous image. There was only one /var/log which was never overflowing.
Also, the /var/log directory gets full very fast.
Is this an config error and how to ensure that all the logs land on this (big) hdd

[image]

diglos, 2021-04-03:
Hi, thanks for reporting, we'll take a look ASAP.

Yes, we included the Armbian approach to deal with /var/log in order to save some MicroSD lifetime (it is now writing on a RAM partition and archiving in the MicroSD).

handelaar2, 2021-04-06:
Thanks Diego. Running  `sudo journalctl -f` looks like the picture below. flooded with write errors. The Eth1-node seems to be running  fine. 
[image]

washosk, 2021-04-06:
Hi! 
we set a default of 200Mb for the RAM log folder. Probably in your installation is not enough, please can you increase the folder size to 400M in the file _/etc/default/armbian-ramlog_

`SIZE=400M
`
And restart the service

`systemctl restart armbian-ramlog
`
I think that it could resolve your issue. Probably in next revisions we also increase this limit.

diglos, 2021-04-07:
Hi.

There was a permission issue with the script that truncates the logs. It should be fixed now. Can you update the package and report back if it works for you?

sudo apt-get update && sudo apt-get install ethereumonarm-armbian-extras

handelaar2, 2021-04-07:
Increasing the default to 400M (as Fernando proposed) already brought some relieve. 
I just ran the package upgrade.
It will take some time for the logs to fill. I did however notice that the default size in the updated package is set to 50M. Fernando mentioned that the default would be 200M. So there is a mismatch there. I left the default to the increased value of 400M

[image]

diglos, 2021-04-07:
Hi,

Yes, Fernando mentioned that before we noticed that something was wrong with the script permissions. It is better to keep it with a tiny size (in order to save RAM) as the script takes care of truncating the logs and preventing it from running out of space.

When the size reaches 75% of the default size the script moves the logs from /var/log (RAM partition) to /var/log.hdd (mounted on the MicroSD) and truncates the logs freeing /var/log space and archiving them to /var/log.hdd. This saves the lifespan of the MicroSD as it doesn't take all log writes.

handelaar2, 2021-04-08:
looks good to me now. Thanks!

handelaar2, 2021-04-15:
After 7 days of no issues, the /var/log directory has become full again. It seems to me there is still some kind of problem.
(what can also be seen is that is it slightly stepping higher, from 25% to almost 50% before it hits 100%)
[image]

washosk, 2021-04-17:
Hi @handelaar2,
seems like some log grew exponentially faster than we expected, and the cron script who moves the data to log.hdd didn't run  in time. In our rpi it never happened so please could you send us the log sizes of the var/log folder.

`sudo du -sh /var/log/*`

as a workarround manually running the truncate-log script and restarting rsyslog shoudl solve it

```
sudo  /usr/lib/armbian/armbian-truncate-logs
sudo systemctl restart rsyslog
```

handelaar2, 2021-04-17:
output of sudo du -sh /var/log/*
/var/log/journal seems very big (96 files of 2.5M)

`4.0K	/var/log/alternatives.log`
`36K	/var/log/apt`
`0	/var/log/armbian-hardware-monitor.log`
`12K	/var/log/auth.log`
`44K	/var/log/auth.log.1`
`8.0K	/var/log/boot.log`
`8.0K	/var/log/boot.log.1`
`8.0K	/var/log/boot.log.2`
`8.0K	/var/log/boot.log.3`
`8.0K	/var/log/boot.log.4`
`8.0K	/var/log/boot.log.5`
`8.0K	/var/log/boot.log.6`
`8.0K	/var/log/boot.log.7`
`0	/var/log/btmp`
`60K	/var/log/cloud-init-output.log`
`1.6M	/var/log/cloud-init.log`
`4.0K	/var/log/dist-upgrade`
`44K	/var/log/dmesg`
`44K	/var/log/dmesg.0`
`12K	/var/log/dpkg.log`
`1.7M	/var/log/grafana`
`243M	/var/log/journal`
`352K	/var/log/kern.log`
`348K	/var/log/kern.log.1`
`4.0K	/var/log/landscape`
`292K	/var/log/lastlog`
`16K	/var/log/lost+found`
`4.0K	/var/log/private`
`4.0K	/var/log/prometheus`
`51M	/var/log/syslog`
`52M	/var/log/syslog.1`
`6.3M	/var/log/syslog.2.gz`
`6.0M	/var/log/syslog.3.gz`
`5.5M	/var/log/syslog.4.gz`
`5.0M	/var/log/syslog.5.gz`
`4.5M	/var/log/syslog.6.gz`
`4.0M	/var/log/syslog.7.gz`
`72K	/var/log/unattended-upgrades`
`164K	/var/log/wtmp`

diglos, 2021-04-19:
Hi, could you run this command and paste here the output?:

ls -l /etc/cron.d/armbian-truncate-logs

Thanks!

handelaar2, 2021-04-20:
output of the ls command:
`-rw-rw-r-- 1 root root 125 Jan 30 10:16 /etc/cron.d/armbian-truncate-logs`

diglos, 2021-04-20:
Yeah, the truncate script has the wrong permissions. Please run:

`sudo chmod 644 /etc/cron.d/armbian-truncate-logs`

This should fix the issue. You can manually run the truncate script in order to free /var/log space:

`sudo /usr/lib/armbian/armbian-truncate-logs`

handelaar2, 2021-04-20:
After running the chmod, the file has following
(write from group has gone, but no +x . I assume that is ok)
`-rw-r--r-- 1 root root 125 Jan 30 10:16 /etc/cron.d/armbian-truncate-logs`

Fernando already mentioned that script, so I was using it already to clean things up.

Muchos gracias Diego for all your hard work!!

diglos, 2021-04-20:
Great, it is fine now. It should truncate logs automatically.

diglos, 2021-04-22:
Hi, please can you confirm that it works now?

Thanks!

handelaar2, 2021-04-22:
Yes, seems ok. No more issue. Thanks!