EOA-Blockchain-Labs/ethereumonarm #4: Failed to start geth

wongtam, 2021-04-22:
Hi! I haven't been able to get my raspberry pi 4 up and running, The follow is what I get when I check the logs upon start up.  I am following the instructions on the readthedocs so not sure what's wrong. Any help would be appreciated...thanks!

-- Logs begin at Wed 2020-04-01 17:23:43 UTC. --
Apr 22 03:55:33 ethereumonarm-37ed4dd2b geth[2725]: INFO [04-22|03:55:33.087] Maximum peer count                       ETH=50 LES=0 total=50
Apr 22 03:55:33 ethereumonarm-37ed4dd2b geth[2725]: INFO [04-22|03:55:33.087] Smartcard socket not found, disabling    err="stat /run/pcscd/pcscd.comm: no such file or directory"
Apr 22 03:55:33 ethereumonarm-37ed4dd2b geth[2725]: Fatal: Failed to create the protocol stack: mkdir /home/ethereum: permission denied
Apr 22 03:55:33 ethereumonarm-37ed4dd2b systemd[1]: geth.service: Main process exited, code=exited, status=1/FAILURE
Apr 22 03:55:33 ethereumonarm-37ed4dd2b systemd[1]: geth.service: Failed with result 'exit-code'.
Apr 22 03:55:33 ethereumonarm-37ed4dd2b systemd[1]: geth.service: Scheduled restart job, restart counter is at 5.
Apr 22 03:55:33 ethereumonarm-37ed4dd2b systemd[1]: Stopped Ethereum geth client daemon by The Ethereum Foundation.
Apr 22 03:55:33 ethereumonarm-37ed4dd2b systemd[1]: geth.service: Start request repeated too quickly.
Apr 22 03:55:33 ethereumonarm-37ed4dd2b systemd[1]: geth.service: Failed with result 'exit-code'.
Apr 22 03:55:33 ethereumonarm-37ed4dd2b systemd[1]: Failed to start Ethereum geth client daemon by The Ethereum Foundation.

diglos, 2021-04-22:
Hi,

It appears that something went wrong with the disk setup (the USB SSD disk). Please run these commands and paste the output here:

`df -h`

`sudo fdisk -l`

Regards.

wongtam, 2021-04-22:
I did a bit more research, I think it might be this issue:
https://github.com/raspberrypi/linux/issues/4130
I am using the same SSD case as in the link.  Will experiment and see If I can get more power to it. Thanks.

diglos, 2021-04-22:
Yes, it seems the NVMe to USB case is not working. I'm closing the issue for now but feel free to reopen it if you need further assistance.

Regards!

closed 2021-04-22 by diglos