EOA-Blockchain-Labs/ethereumonarm #8: Incorrect default `geth` config

trevor-scheer, 2021-05-15:
Hi, I've followed the tutorial and used the downloads from [this page](https://ethereum.org/en/developers/tutorials/run-node-raspberry-pi/) to set up my rpi4 as an Eth 2.0 validator.

I noticed that `prysm-beacon` was logging errors about not being able to connect to eth1, even though `geth` was running - I even waited for it to finish syncing. The solution for me was to edit the `geth.conf` file to include the `--http` flag and restart the service. It seems this argument should already be there by default.

trevor-scheer, 2021-05-15:
It seems the image on the ethereum.org website is dated, since the `--http` flag has clearly [existed since the config was introduced](https://github.com/diglos/ethereumonarm/commit/511cbd1ef997cfd78efc4f5435aa9002669c56fc#diff-a99686e22ed7e972776291175ebfa22b21a48ebaee960161c394443847f925fb) within this repo.

I'm curious if the article on ethereum.org should just link out to [the actual ethereum-on-arm docs](https://ethereum-on-arm-documentation.readthedocs.io/) instead.

diglos, 2021-05-17:
Hi,

yes, your are right, I will contact EF to try to get the link updated. I'm leaving this issue opened until this is fixed.

Thanks!

trevor-scheer, 2021-05-17:
Thanks @diglos!

closed 2025-06-19 by trevor-scheer