EOA-Blockchain-Labs/ethereumonarm #10: Seems a pi with geth no longer manages to sync

cryptogohan, 2021-05-28:
It's been a month. Recommended hardware (4gb pi + active cooling + 1TB ssd + 200mbps connection). Block headers look good, state trie looks neverending.
```
May 28 09:42:36 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:42:36.100] Imported new block receipts              count=64   elapsed=661.907ms   number=12522015 hash="b2018b…8c6303" age=15m8s    size=7.83MiB
May 28 09:42:38 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:42:38.423] Imported new state entries               count=768  elapsed=29.046ms    processed=1001037433 pending=40452  trieretry=2    coderetry=0 duplicate=763 unexpected=3267
May 28 09:42:39 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:42:39.245] Imported new state entries               count=273  elapsed="7.204µs"   processed=1001037706 pending=4319   trieretry=0    coderetry=0 duplicate=763 unexpected=3267
May 28 09:42:41 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:42:41.265] Imported new state entries               count=384  elapsed="7.389µs"   processed=1001038090 pending=5682   trieretry=0    coderetry=0 duplicate=763 unexpected=3267
May 28 09:42:43 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:42:43.084] Imported new state entries               count=384  elapsed="7.351µs"   processed=1001038474 pending=7010   trieretry=0    coderetry=0 duplicate=763 unexpected=3267
May 28 09:42:44 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:42:44.797] Imported new state entries               count=384  elapsed="7.889µs"   processed=1001038858 pending=8299   trieretry=0    coderetry=0 duplicate=763 unexpected=3267
May 28 09:42:46 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:42:46.395] Imported new state entries               count=384  elapsed="14.722µs"  processed=1001039242 pending=9598   trieretry=0    coderetry=0 duplicate=763 unexpected=3267
May 28 09:42:47 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:42:47.854] Imported new state entries               count=384  elapsed="7.351µs"   processed=1001039626 pending=10935  trieretry=0    coderetry=0 duplicate=763 unexpected=3267
May 28 09:42:48 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:42:48.331] Imported new state entries               count=206  elapsed="7.445µs"   processed=1001039832 pending=14220  trieretry=0    coderetry=0 duplicate=763 unexpected=3267
May 28 09:42:49 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:42:49.662] Imported new state entries               count=384  elapsed="8.315µs"   processed=1001040216 pending=15526  trieretry=0    coderetry=0 duplicate=763 unexpected=3267
May 28 09:42:50 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:42:50.965] Imported new state entries               count=384  elapsed="7.63µs"    processed=1001040600 pending=17461  trieretry=0    coderetry=0 duplicate=763 unexpected=3267
May 28 09:42:51 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:42:51.884] Imported new state entries               count=384  elapsed="7.129µs"   processed=1001040984 pending=23605  trieretry=0    coderetry=0 duplicate=763 unexpected=3267
May 28 09:42:52 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:42:52.728] Imported new state entries               count=384  elapsed="17.926µs"  processed=1001041368 pending=29749  trieretry=0    coderetry=0 duplicate=763 unexpected=3267
May 28 09:42:53 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:42:53.633] Imported new state entries               count=384  elapsed="7.999µs"   processed=1001041752 pending=35893  trieretry=0    coderetry=0 duplicate=763 unexpected=3267
May 28 09:42:57 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:42:57.977] Imported new state entries               count=384  elapsed="6.314µs"   processed=1001042136 pending=36323  trieretry=0    coderetry=0 duplicate=763 unexpected=3267
May 28 09:43:01 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:43:01.682] Imported new state entries               count=384  elapsed="15.463µs"  processed=1001042520 pending=36767  trieretry=0    coderetry=0 duplicate=763 unexpected=3267
May 28 09:43:05 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:43:05.267] Imported new state entries               count=384  elapsed="10.315µs"  processed=1001042904 pending=37183  trieretry=0    coderetry=0 duplicate=763 unexpected=3267
May 28 09:43:08 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:43:08.709] Imported new state entries               count=384  elapsed="7.444µs"   processed=1001043288 pending=37612  trieretry=0    coderetry=0 duplicate=763 unexpected=3267
May 28 09:43:12 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:43:12.627] Imported new state entries               count=384  elapsed="19.796µs"  processed=1001043672 pending=38037  trieretry=0    coderetry=0 duplicate=763 unexpected=3267
May 28 09:43:16 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:43:16.004] Imported new state entries               count=384  elapsed=5.400ms     processed=1001044056 pending=38467  trieretry=0    coderetry=0 duplicate=763 unexpected=3267
May 28 09:43:19 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:43:19.289] Imported new state entries               count=384  elapsed="7.741µs"   processed=1001044440 pending=38902  trieretry=0    coderetry=0 duplicate=763 unexpected=3267
May 28 09:43:22 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:43:22.656] Imported new state entries               count=384  elapsed="22.315µs"  processed=1001044824 pending=39330  trieretry=0    coderetry=0 duplicate=763 unexpected=3267
May 28 09:43:25 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:43:25.989] Imported new state entries               count=384  elapsed="7.666µs"   processed=1001045208 pending=39763  trieretry=0    coderetry=0 duplicate=763 unexpected=3267
May 28 09:43:27 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:43:27.867] Imported new block headers               count=1    elapsed=551.826ms   number=12522079 hash="793681…ca082d" age=1m28s
May 28 09:43:28 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:43:28.567] Downloader queue stats                   receiptTasks=0    blockTasks=0    itemSize=252.00KiB throttle=261
May 28 09:43:29 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:43:29.516] Imported new state entries               count=384  elapsed="162.461µs" processed=1001045592 pending=40192  trieretry=0    coderetry=0 duplicate=763 unexpected=3267
May 28 09:43:30 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:43:30.528] Imported new block headers               count=1    elapsed=16.692ms    number=12522080 hash="322704…e31f18" age=1m21s
May 28 09:43:32 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:43:32.874] Imported new state entries               count=384  elapsed="131.165µs" processed=1001045976 pending=40608  trieretry=0    coderetry=0 duplicate=763 unexpected=3267
May 28 09:43:33 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:43:33.728] Imported new block headers               count=1    elapsed=13.803ms    number=12522081 hash="ed0a88…a9d18d" age=1m1s
May 28 09:43:36 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:43:36.335] Imported new state entries               count=384  elapsed="7.408µs"   processed=1001046360 pending=41032  trieretry=4    coderetry=0 duplicate=763 unexpected=3267
May 28 09:43:45 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:43:45.146] Imported new state entries               count=384  elapsed="222.052µs" processed=1001046744 pending=41413  trieretry=2    coderetry=0 duplicate=763 unexpected=3267
May 28 09:43:46 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:43:46.762] Imported new block headers               count=1    elapsed=540.141ms   number=12522082 hash="945f0b…d57a3b"
May 28 09:43:54 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:43:54.626] Imported new state entries               count=384  elapsed="364.088µs" processed=1001047128 pending=41792  trieretry=2    coderetry=0 duplicate=763 unexpected=3267
May 28 09:43:55 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:43:55.720] Imported new block headers               count=1    elapsed=85.125ms    number=12522083 hash="f942ce…066d95"
May 28 09:44:04 ethereumonarm-2d3f714f1 geth[451683]: INFO [05-28|09:44:04.028] Imported new state entries               count=384  elapsed="15.389µs"  processed=1001047512 pending=42181  trieretry=0    coderetry=0 duplicate=763 unexpected=3267
[... cut ...]
Going to try OpenEthereum now. From what I understand it wouldn't get stuck on the state trie. Shame that there's no Grafana dashboard, maybe I can enable metrics and pluck one from somewhere.

This issue can be closed, just thought it be good to report in case others try setting up a Pi + Geth and find they're still not synched weeks later.

Perhaps someone also has suggestions for how to get the pi synched. Running with default arguments:
```
ExecStart=/usr/bin/geth --http --http.port 8545 --http.addr 0.0.0.0 --http.corsdomain '*' --cache 2048 --metrics --metrics.expensive --pprof --http --snapshot=false
```

tmuras, 2021-07-03:
I had synch problems as well, playing with the options / restarting didn't help.
What did help was the manual installation of the geth software from https://geth.ethereum.org/downloads/

diglos, 2021-07-05:
Hi,

Can you give us more info on your setup? (RAM, SSD, etc). We will enable the snap sync on the next release which should make it easy get the node synced. 

With fast sync and an 8 GB RAM Raspberry Pi (with a decent SSD) you should get it in sync in 2-3 days. (State entries can go as far as 800 Million today).

tmuras, 2021-07-05:
By looking at the log and measuring the time I came up with the current speed of 6000 state entries processed per minute.
It would mean that 800 Million will take 92 days.

However, I'm currently at 700 millions and the machine is running for 6 days. So it may be that it's just slowing down.

I'm using Raspberry Pi model B with 8GB of RAM.
The SSD is GoodRam CL100 Gen3 960 GB 2.5" SATA III (SSDPR-CL100-960-G3). It's connected with USB 3, so it may be that the connector is causing performance loss.
In https://ethereum-on-arm-documentation.readthedocs.io/en/latest/user-guide/storage.html you mention:

"Use an USB 3.0 External Hard Drive Case with a SSD Disk. In our case we used a Inateck 2.5 Hard Drive Enclosure FE2011. Make sure to buy a case with an UAS (USB Attached SCSI) compliant chip..."

Would you know how to check if my enclosure is UAS compliant or not?

`hdparm -I /dev/sda

/dev/sda:

ATA device, with non-removable media
	Model Number:       SSDPR-CL100-960-G3                      
	Serial Number:      XY2141635           
	Firmware Revision:  S1127B0 
	Media Serial Num:   
	Media Manufacturer: 
	Transport:          Serial, ATA8-AST, SATA 1.0a, SATA II Extensions, SATA Rev 2.5, SATA Rev 2.6, SATA Rev 3.0
Standards:
	Used: unknown (minor revision code 0x0110) 
	Supported: 9 8 7 6 5 
	Likely used: 9
Configuration:
	Logical		max	current
	cylinders	16383	16383
	heads		16	16
	sectors/track	63	63
	--
	CHS current addressable sectors:    16514064
	LBA    user addressable sectors:   268435455
	LBA48  user addressable sectors:  1875385008
	Logical  Sector size:                   512 bytes
	Physical Sector size:                   512 bytes
	Logical Sector-0 offset:                  0 bytes
	device size with M = 1024*1024:      915715 MBytes
	device size with M = 1000*1000:      960197 MBytes (960 GB)
	cache/buffer size  = unknown
	Form Factor: 2.5 inch
	Nominal Media Rotation Rate: Solid State Device
Capabilities:
	LBA, IORDY(can be disabled)
	Queue depth: 32
	Standby timer values: spec'd by Standard, no device specific minimum
	R/W multiple sector transfer: Max = 1	Current = 1
	Advanced power management level: 128
	DMA: mdma0 mdma1 mdma2 udma0 udma1 udma2 udma3 udma4 udma5 *udma6 
	     Cycle time: min=120ns recommended=120ns
	PIO: pio0 pio1 pio2 pio3 pio4 
	     Cycle time: no flow control=120ns  IORDY flow control=120ns
Commands/features:
	Enabled	Supported:
	   *	SMART feature set
	    	Security Mode feature set
	   *	Power Management feature set
	   *	Write cache
	   *	Look-ahead
	   *	Host Protected Area feature set
	   *	WRITE_BUFFER command
	   *	READ_BUFFER command
	   *	DOWNLOAD_MICROCODE
	   *	Advanced Power Management feature set
	    	SET_MAX security extension
	   *	48-bit Address feature set
	   *	Mandatory FLUSH_CACHE
	   *	FLUSH_CACHE_EXT
	   *	SMART error logging
	   *	SMART self-test
	   *	General Purpose Logging feature set
	   *	WRITE_{DMA|MULTIPLE}_FUA_EXT
	   *	WRITE_UNCORRECTABLE_EXT command
	   *	{READ,WRITE}_DMA_EXT_GPL commands
	   *	Segmented DOWNLOAD_MICROCODE
	   *	Gen1 signaling speed (1.5Gb/s)
	   *	Gen2 signaling speed (3.0Gb/s)
	   *	Gen3 signaling speed (6.0Gb/s)
	   *	Native Command Queueing (NCQ)
	   *	Phy event counters
	   *	READ_LOG_DMA_EXT equivalent to READ_LOG_EXT
	    	DMA Setup Auto-Activate optimization
	   *	Software settings preservation
	   *	SANITIZE feature set
	   *	BLOCK_ERASE_EXT command
	   *	DOWNLOAD MICROCODE DMA command
	   *	WRITE BUFFER DMA command
	   *	READ BUFFER DMA command
	   *	Data Set Management TRIM supported (limit 8 blocks)
	   *	Deterministic read ZEROs after TRIM
Security: 
	Master password revision code = 65534
		supported
	not	enabled
	not	locked
	not	frozen
	not	expired: security count
		supported: enhanced erase
	2min for SECURITY ERASE UNIT. 2min for ENHANCED SECURITY ERASE UNIT.
Checksum: correct
`

diglos, 2021-07-05:
Could you run:

`lsusb -v`

And paste it here?

tmuras, 2021-07-06:
```
ethereum@ethnode-81702b553:~$ sudo lsusb -v
[sudo] password for ethereum: 

Bus 002 Device 002: ID 11b0:6388 ATECH FLASH TECHNOLOGY SNA-DCH/U3
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         9
  idVendor           0x11b0 ATECH FLASH TECHNOLOGY
  idProduct          0x6388 
  bcdDevice           54.08
  iManufacturer           4 Kingston
  iProduct                5 SNA-DCH/U3
  iSerial                 6 0000000000019834
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x002c
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              896mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              7 SATA
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x0016
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
[... 248 lines cut ...]
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x02
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0507 highspeed power suspend enable connect
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

diglos, 2021-07-07:
Ummh, Kingston. Based on our experience, it's performance is terrible with the Rpis. I'm not saying I'm 100% sure this is the problem but it is the most probable cause.

You can do an iozone3 disk test to check the disk performance (but you need to stop geth service first).

```
sudo apt-get install iozone3
iozone -e -I -a -s 1000M -r 4k -r 16k -r 128k -r 512k -r 1024k -r 16384k -i 0 -i 1 -i 2
```

Paste the results here.

alextes, 2021-07-07:
(I'm the alt of cryptogohan. Transitioning to this handle.)

Yea, I'm on the 4gb model I'm afraid. I suppose I'll just sell it then. I got the recommended Samsung SSD I think.

`lsusb -v`
```
Bus 002 Device 002: ID 04e8:61f5 Samsung Electronics Co., Ltd Portable SSD T5
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         9
  idVendor           0x04e8 Samsung Electronics Co., Ltd
  idProduct          0x61f5 Portable SSD T5
  bcdDevice            1.00
  iManufacturer           2 Samsung
  iProduct                3 Portable SSD T5
  iSerial                 1 123456871E98
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0079
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              896mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           4
[... 331 lines cut ...]
  nNbrPorts             1
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x02
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0507 highspeed power suspend enable connect
Device Status:     0x0001
  Self Powered
```

Latest geth logs:
```
Jul 07 16:07:33 ethereumonarm-2d3f714f1 geth[541969]: INFO [07-07|16:07:33.291] Imported new state entries               count=1153 elapsed=90.423ms     processed=1079508707 pending=96015  trieretry=0   coderetry=0 duplicate=359 unexpected=2401
Jul 07 16:07:34 ethereumonarm-2d3f714f1 geth[541969]: INFO [07-07|16:07:34.607] Imported new state entries               count=1152 elapsed=91.814ms     processed=1079509859 pending=95005  trieretry=0   coderetry=0 duplicate=359 unexpected=2401
Jul 07 16:07:35 ethereumonarm-2d3f714f1 geth[541969]: INFO [07-07|16:07:35.861] Imported new state entries               count=1152 elapsed=88.711ms     processed=1079511011 pending=93760  trieretry=0   coderetry=0 duplicate=359 unexpected=2401
Jul 07 16:07:37 ethereumonarm-2d3f714f1 geth[541969]: INFO [07-07|16:07:37.268] Imported new state entries               count=1152 elapsed=96.559ms     processed=1079512163 pending=92479  trieretry=0   coderetry=0 duplicate=359 unexpected=2401
Jul 07 16:07:41 ethereumonarm-2d3f714f1 geth[541969]: INFO [07-07|16:07:41.487] Imported new state entries               count=1153 elapsed=97.062ms     processed=1079513316 pending=91731  trieretry=0   coderetry=0 duplicate=359 unexpected=2401
Jul 07 16:07:45 ethereumonarm-2d3f714f1 geth[541969]: INFO [07-07|16:07:45.334] Imported new state entries               count=768  elapsed=48.982ms     processed=1079514084 pending=91075  trieretry=0   coderetry=0 duplicate=359 unexpected=2401
Jul 07 16:07:48 ethereumonarm-2d3f714f1 geth[541969]: INFO [07-07|16:07:48.827] Imported new state entries               count=768  elapsed=47.488ms     processed=1079514852 pending=90211  trieretry=0   coderetry=0 duplicate=359 unexpected=2401
```

tmuras, 2021-07-21:
In my case the problem was with the external SSD enclosure. It was causing slow-down and even IO errors. Once I've replaced it with the new one, the synchronization finishes.
Here is the last entry about processing state entries from my log. The entry is from July 15 2021:

```
Imported new state entries               count=1558 elapsed=86.216ms     processed=926,925,878 pending=0       trieretry=0    coderetry=0 duplicate=4753 unexpected=20735
```

tmuras, 2021-07-21:
I have tried a different approach and used RockPi with 4GB RAM but NVMe attached directly to it.
I took latest 1.10.4-stable-aa637fd3 .

The synchronization was quick quick (about a week) and the node catch up with new blocks:

```
> eth.syncing
{ 
  currentBlock: 12865866,
  highestBlock: 12867983,
  knownStates: 0,
  pulledStates: 0,
  startingBlock: 12864834
}

```

But I guess this is not over, because web3.eth.getBalance still returns 0.

What I see in the logs is now "State heal in progress":

```
INFO [07-21|07:39:10.849] State heal in progress                   accounts=4,809,912@250.08MiB slots=7,283,188@546.00MiB codes=2736@21.77MiB nodes=62,716,384@18.27GiB pending=133,461
```

This has been running for over a week now.
Once I've restarted blockchain, it seems that this heal process starts from the beginning:

```
INFO [07-21|07:43:47.891] State heal in progress                   accounts=0@0.00B slots=0@0.00B codes=2736@21.77MiB nodes=62,719,677@18.27GiB pending=30550
```

Would you know what are accounts, slots, codes, nodes and pending there?

diglos, 2021-07-21:
Yeah, it is not over, healing is part of the new snap sync mode so it will take some time.

4 GB is really tight, please let us know if it finishes the sync and how long it takes.

Thanks!

PierrickGT, 2021-08-12:
I have a similar issue with my setup, the sync is stopping at around 70% with the error: `Database compacting, degraded performance`

Iozone test:
```
ethereum@ethereumonarm-9c861e5df:~$ iozone -e -I -a -s 1000M -r 4k -r 16k -r 128k -r 512k -r 1024k -r 16384k -i 0 -i 1 -i 2
	Iozone: Performance Test of File I/O
	        Version $Revision: 3.489 $
		Compiled for 64 bit mode.
		Build: linux

	Contributors:William Norcott, Don Capps, Isom Crawford, Kirby Collins
	             Al Slater, Scott Rhine, Mike Wisner, Ken Goss
	             Steve Landherr, Brad Smith, Mark Kelly, Dr. Alain CYR,
	             Randy Dunlap, Mark Montague, Dan Million, Gavin Brebner,
	             Jean-Marc Zucconi, Jeff Blomberg, Benny Halevy, Dave Boone,
	             Erik Habbinga, Kris Strecker, Walter Wong, Joshua Root,
	             Fabrice Bacchella, Zhenghua Xue, Qin Li, Darren Sawyer,
	             Vangel Bojaxhi, Ben England, Vikentsi Lapa,
	             Alexey Skidanov, Sudhir Kumar.

	Run began: Thu Aug 12 03:28:23 2021

	Include fsync in write timing
	O_DIRECT feature enabled
	Auto Mode
	File size set to 1024000 kB
	Record Size 4 kB
	Record Size 16 kB
	Record Size 128 kB
	Record Size 512 kB
	Record Size 1024 kB
	Record Size 16384 kB
	Command line used: iozone -e -I -a -s 1000M -r 4k -r 16k -r 128k -r 512k -r 1024k -r 16384k -i 0 -i 1 -i 2
	Output is in kBytes/sec
	Time Resolution = 0.000001 seconds.
	Processor cache size set to 1024 kBytes.
	Processor cache line size set to 32 bytes.
	File stride size set to 17 * record size.
                                                              random    random     bkwd    record    stride
              kB  reclen    write  rewrite    read    reread    read     write     read   rewrite      read   fwrite frewrite    fread  freread
         1024000       4    23100    28344    28479    28539    19739    27929
         1024000      16    69194    80482    91274    91784    48724    79205
         1024000     128   181450   184068   185135   184869   127432   184495
         1024000     512   199708   203982   206288   206141   194226   205738
         1024000    1024   232948   238272   256269   258775   246168   238388
         1024000   16384   298318   301506   339673   345170   334623   301269

iozone test complete.
```

And here is the `lsusb -v` report:

```
ethereum@ethereumonarm-9c861e5df:~$ lsusb -v

Bus 002 Device 002: ID 174c:235c ASMedia Technology Inc. Ugreen Storage Device
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.10
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         9
  idVendor           0x174c ASMedia Technology Inc.
  idProduct          0x235c
  bcdDevice            1.00
  iManufacturer           2
  iProduct                3
[... 233 lines cut ...]
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
        ```
        
I'm running it on a 8Gb RPI4 version, so I suppose the problem is either coming from the enclosure or the cable, when I look at grafana, I can see that the max writing speed is around 30 MB/s but usually it's writing at 20 MB/s so I wonder if it could be the root of the issue.

Any SSD enclosure to recommend?

diglos, 2021-09-21:
Hi, sorry I missed this comment. 

`Database compacting, degraded performance` is not an error, is a common task during sync.

20/30MB/s are expected figures (as there a lot of small read/writes). Iozone shows that performance is good.

Are you doing ok with the node?

PierrickGT, 2021-09-24:
> Hi, sorry I missed this comment.
> 
> `Database compacting, degraded performance` is not an error, is a common task during sync.
> 
> 20/30MB/s are expected figures (as there a lot of small read/writes). Iozone shows that performance is good.
> 
> Are you doing ok with the node?

Thanks for replying! 
I sent back the SSD enclosure since I resynced my node a second time and the sync was still stuck on the `Database compacting, degraded performance` message even after a couple days.
I will get a new SSD enclosure and try again.
Thanks for your help.

closed 2025-07-04 by washosk