EOA-Blockchain-Labs/ethereumonarm #12: Slow SSH logins

luisnaranjo733, 2021-06-30:
I noticed that SSH logins are excruciatingly slow with this Ubuntu server image. 

I did some troubleshooting by passing the -vv flag to ssh to see what it was getting hung up on, and it was getting stuck on "pledge: network".

This is caused by an optional pluggable authentication module (PAM).
https://serverfault.com/a/998761

Commenting out 
> session optional pam_systemd.so

in /etc/pam.d/common-session solves this slow SSH login issue. This might be worth adding to the rc.local via a sed command

Sources:
- https://serverfault.com/a/998761
- https://linuxize.com/post/how-to-configure-static-ip-address-on-ubuntu-20-04/

luisnaranjo733, 2021-06-30:
> sed -i 's|session\soptional\s\bpam_systemd.so|#session optional        pam_systemd.so|g' /etc/pam.d/common-session

washosk, 2021-07-04:
I will take a deeper look into this, bc in our test environment we never suffer this slowness, give my a few days to learn more about this and do further testing.