EOA-Blockchain-Labs/ethereumonarm #13: Swap file not getting mounted automatically

luisnaranjo733, 2021-07-20:
Testing out the ```thereumonarm-armbian-extra``` package and I noticed that the swap file that gets set up by the postinstall script doesn't get mounted after a reboot. Running the ```systemctl enable dphys-swapfile``` command manually seems to bring it back up.

```bash

~$ sudo cat /proc/swaps
Filename				Type		Size	Used	Priority
/dev/zram0                              partition	1599744	0	5

~$ sudo systemctl enable dphys-swapfile
Synchronizing state of dphys-swapfile.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable dphys-swapfile

~$ sudo cat /proc/swaps
Filename				Type		Size	Used	Priority
/dev/zram0                              partition	1599744	0	5
/home/ethereum/swapfile                 file		8388604	0	-2
```

Has anyone done any analysis on how often you burn through ZRAM and end up using swap? I would hope never, but I'm just curious

Also curious about the rationale for splicing the armbian ZRAM config into a custom package. What advantages/disadvantages do you see vs a pre-packaged zram swap script like [this one](https://github.com/foundObjects/zram-swap)?

luisnaranjo733, 2021-07-20:
Hmm this actually might be a fluke. I just did another reboot and this time the swapfile was mounted automatically without needing to run ```systemctl enable dphys-swapfile``` manually.
Feel free to close this if you can't reproduce

diglos, 2021-07-21:
Hi, Ok.

Regarding ZRAM, Armbian is focused on optimizing constrained-resourced devices and particularly the ones that use MicroSD cards. please see this note by TKaiser:

https://forum.armbian.com/topic/5565-zram-vs-swap/?do=findComment&comment=62574

The whole thread is very interesting as well:

https://forum.armbian.com/topic/5565-zram-vs-swap/

closed 2021-07-21 by diglos