EOA-Blockchain-Labs/ethereumonarm #14: State heal in progress

nimcon, 2021-07-20:
Trying to get a Geth node to run on a RP4 with 8GB RAM and 2 TB USB 3.0 SSD. However, after the state sync is completed, it goes straight into a state heal. Is this expected, and if so, how long will this phase take? If it's not expected, what went wrong and how to fix?

```Jul 19 19:56:15 ethereumonarm-f93a22f29 geth[79761]: INFO [07-19|19:56:15.951] Imported new block headers               count=1    elapsed=13.469ms   number=12,859,016 hash=2d82c2..2cf337
Jul 19 19:56:17 ethereumonarm-f93a22f29 geth[79761]: INFO [07-19|19:56:17.276] State sync in progress                   synced=99.98% state=131.91GiB  accounts=159,411,374@32.46GiB slots=486,178,145@97.28GiB codes=418,259@2.18GiB    eta=-1m55.188s
Jul 19 19:56:17 ethereumonarm-f93a22f29 geth[79761]: WARN [07-19|19:56:17.964] Checkpoint challenge timed out, dropping id=71f102b6bfe7d8c8 conn=inbound addr=54.73.15.74:51648                 type=OpenEthereum/v3.2.6-...
Jul 19 19:56:20 ethereumonarm-f93a22f29 geth[79761]: INFO [07-19|19:56:20.647] Imported new block receipts              count=678  elapsed=5.601s     number=12,858,945 hash=f629cd..396da8 age=15m9s      size=78.64MiB
Jul 19 19:56:21 ethereumonarm-f93a22f29 geth[79761]: INFO [07-19|19:56:21.351] State sync in progress                   synced=99.99% state=131.92GiB  accounts=159,433,233@32.46GiB slots=486,194,040@97.28GiB codes=418,316@2.18GiB    eta=-2m9.177s
Jul 19 19:56:21 ethereumonarm-f93a22f29 geth[79761]: WARN [07-19|19:56:21.616] Unexpected account range packet          peer=27ddc68d         reqid=4,939,331,682,164,732,091
Jul 19 19:56:24 ethereumonarm-f93a22f29 geth[79761]: INFO [07-19|19:56:24.224] State sync in progress                   synced=100.00% state=131.93GiB  accounts=159,441,881@32.46GiB slots=486,198,626@97.28GiB codes=418,346@2.18GiB    eta=-2m16.058s
Jul 19 19:56:24 ethereumonarm-f93a22f29 geth[79761]: WARN [07-19|19:56:24.863] Checkpoint challenge timed out, dropping id=9c6767a3e0f65bc7 conn=dyndial addr=208.59.38.47:30303                type=Geth/v1.10.5-stable-...
Jul 19 19:56:37 ethereumonarm-f93a22f29 geth[79761]: WARN [07-19|19:56:37.088] Checkpoint challenge timed out, dropping id=f1f3099a705be65f conn=inbound addr=186.22.18.201:56358               type=erigon/v2021.07.3-al...
Jul 19 19:56:40 ethereumonarm-f93a22f29 geth[79761]: INFO [07-19|19:56:40.564] State heal in progress                   accounts=0@0.00B              slots=0@0.00B              codes=0@0.00B            nodes=370@192.23KiB pending=5609
Jul 19 19:56:40 ethereumonarm-f93a22f29 geth[79761]: WARN [07-19|19:56:40.571] Unexpected trienode heal packet          peer=9fccc38c         reqid=2,779,127,575,212,821,235
Jul 19 19:56:40 ethereumonarm-f93a22f29 geth[79761]: WARN [07-19|19:56:40.617] Unexpected trienode heal packet          peer=9ff17625         reqid=5,925,214,810,693,714,548
Jul 19 19:56:41 ethereumonarm-f93a22f29 geth[79761]: WARN [07-19|19:56:41.949] Unexpected trienode heal packet          peer=e70811b4         reqid=7,690,861,252,952,579,679
Jul 19 19:56:44 ethereumonarm-f93a22f29 geth[79761]: WARN [07-19|19:56:44.035] Unexpected trienode heal packet          peer=2cd761db         reqid=2,756,345,096,628,177,020
Jul 19 19:56:44 ethereumonarm-f93a22f29 geth[79761]: WARN [07-19|19:56:44.068] Unexpected trienode heal packet          peer=73fd19d8         reqid=6,385,141,813,524,173,251
Jul 19 19:56:44 ethereumonarm-f93a22f29 geth[79761]: WARN [07-19|19:56:44.276] Unexpected trienode heal packet          peer=b65d7b0b         reqid=8,375,173,560,166,334,289
Jul 19 19:56:44 ethereumonarm-f93a22f29 geth[79761]: WARN [07-19|19:56:44.303] Unexpected trienode heal packet          peer=402b4b92         reqid=6,243,137,689,026,562,439
Jul 19 19:56:44 ethereumonarm-f93a22f29 geth[79761]: WARN [07-19|19:56:44.487] Unexpected trienode heal packet          peer=fe5f77bf         reqid=7,339,829,303,620,979,336
Jul 19 19:56:44 ethereumonarm-f93a22f29 geth[79761]: WARN [07-19|19:56:44.517] Unexpected trienode heal packet          peer=fd937589         reqid=3,817,543,624,467,065,485
Jul 19 19:56:53 ethereumonarm-f93a22f29 geth[79761]: INFO [07-19|19:56:53.997] State heal in progress                   accounts=0@0.00B              slots=0@0.00B              codes=0@0.00B            nodes=707@367.31KiB pending=10937
Jul 19 19:56:54 ethereumonarm-f93a22f29 geth[79761]: WARN [07-19|19:56:54.000] Unexpected trienode heal packet          peer=2736fd6a         reqid=4,685,317,259,206,057,756
Jul 19 19:56:54 ethereumonarm-f93a22f29 geth[79761]: WARN [07-19|19:56:54.558] Unexpected trienode heal packet          peer=6533a7cc         reqid=4,708,310,855,523,396,133
Jul 19 19:56:55 ethereumonarm-f93a22f29 geth[79761]: WARN [07-19|19:56:55.635] Unexpected trienode heal packet          peer=b639490d         reqid=1,761,834,901,958,553,646```

About fifteen hours later, the log looks like this:

```Jul 20 08:55:57 ethereumonarm-f93a22f29 geth[79761]: WARN [07-20|08:55:57.290] Unexpected trienode heal packet          peer=e03d20ed         reqid=1,377,280,508,779,049,759
Jul 20 08:55:57 ethereumonarm-f93a22f29 geth[79761]: WARN [07-20|08:55:57.294] Unexpected trienode heal packet          peer=61551118         reqid=3,934,290,387,191,793,956
Jul 20 08:55:57 ethereumonarm-f93a22f29 geth[79761]: WARN [07-20|08:55:57.305] Unexpected trienode heal packet          peer=e24ea44e         reqid=1,111,763,378,262,048,808
Jul 20 08:55:57 ethereumonarm-f93a22f29 geth[79761]: WARN [07-20|08:55:57.307] Unexpected trienode heal packet          peer=11d75e71         reqid=2,907,927,302,575,975,489
Jul 20 08:55:57 ethereumonarm-f93a22f29 geth[79761]: WARN [07-20|08:55:57.310] Unexpected trienode heal packet          peer=22e17f9c         reqid=3,516,073,890,914,378,070
Jul 20 08:55:57 ethereumonarm-f93a22f29 geth[79761]: WARN [07-20|08:55:57.311] Unexpected trienode heal packet          peer=cb922008         reqid=1,894,546,731,677,936,354
Jul 20 08:56:01 ethereumonarm-f93a22f29 geth[79761]: INFO [07-20|08:56:01.592] State heal in progress                   accounts=265,245@12.69MiB     slots=209,756@15.56MiB     codes=175@1.35MiB        nodes=2,263,344@696.54MiB pending=91006
Jul 20 08:56:06 ethereumonarm-f93a22f29 geth[79761]: INFO [07-20|08:56:06.285] Imported new block headers               count=1    elapsed=10.008ms   number=12,862,459 hash=45255a..affe22 age=1m45s
Jul 20 08:56:06 ethereumonarm-f93a22f29 geth[79761]: INFO [07-20|08:56:06.862] Downloader queue stats                   receiptTasks=0     blockTasks=0     itemSize=248.19KiB throttle=1057
Jul 20 08:56:10 ethereumonarm-f93a22f29 geth[79761]: INFO [07-20|08:56:10.576] Imported new block headers               count=2    elapsed=13.631ms   number=12,862,461 hash=fd0ef2..f463f1 age=1m27s
Jul 20 08:56:11 ethereumonarm-f93a22f29 geth[79761]: INFO [07-20|08:56:11.143] State heal in progress                   accounts=265,308@12.69MiB     slots=209,897@15.57MiB     codes=175@1.35MiB        nodes=2,263,931@696.65MiB pending=91255
Jul 20 08:56:14 ethereumonarm-f93a22f29 geth[79761]: INFO [07-20|08:56:14.615] Imported new block headers               count=1    elapsed=11.778ms   number=12,862,462 hash=208f43..e6ef2b
Jul 20 08:56:14 ethereumonarm-f93a22f29 geth[79761]: WARN [07-20|08:56:14.867] Pivot became stale, moving               old=12,862,336 new=12,862,400
Jul 20 08:56:15 ethereumonarm-f93a22f29 geth[79761]: INFO [07-20|08:56:15.258] Imported new block receipts              count=64   elapsed=388.930ms  number=12,862,399 hash=e03ec7..423ade age=14m49s     size=7.32MiB
Jul 20 08:56:15 ethereumonarm-f93a22f29 geth[79761]: INFO [07-20|08:56:15.639] State heal in progress                   accounts=265,332@12.69MiB     slots=209,946@15.58MiB     codes=175@1.35MiB        nodes=2,264,160@696.70MiB pending=91439
Jul 20 08:56:15 ethereumonarm-f93a22f29 geth[79761]: WARN [07-20|08:56:15.649] Unexpected trienode heal packet          peer=c8cb5d96         reqid=384,864,180,950,622,558
Jul 20 08:56:15 ethereumonarm-f93a22f29 geth[79761]: WARN [07-20|08:56:15.662] Unexpected trienode heal packet          peer=11d75e71         reqid=3,864,654,568,017,345,329
Jul 20 08:56:15 ethereumonarm-f93a22f29 geth[79761]: WARN [07-20|08:56:15.663] Unexpected trienode heal packet          peer=e24ea44e         reqid=596,666,107,961,133,006
Jul 20 08:56:15 ethereumonarm-f93a22f29 geth[79761]: WARN [07-20|08:56:15.663] Unexpected trienode heal packet          peer=cfe18f98         reqid=2,843,389,484,003,173,002
[... cut ...]
Jul 20 08:56:15 ethereumonarm-f93a22f29 geth[79761]: WARN [07-20|08:56:15.680] Unexpected trienode heal packet          peer=81e8669c         reqid=8,506,724,142,604,709,679
Jul 20 08:56:15 ethereumonarm-f93a22f29 geth[79761]: WARN [07-20|08:56:15.714] Unexpected trienode heal packet          peer=2736fd6a         reqid=7,444,444,972,996,853,561
Jul 20 08:56:15 ethereumonarm-f93a22f29 geth[79761]: WARN [07-20|08:56:15.719] Unexpected trienode heal packet          peer=f21696a2         reqid=6,136,919,150,236,356,465```

diglos, 2021-07-20:
Hi.

Yes, the state heal process is part of the new snap sync. Let us know if it completes the sync.

nimcon, 2021-07-22:
State heal still ongoing. Is it possible to tell how much is left? Here's the latest excerpts from the log:

```Jul 22 13:41:42 ethereumonarm-f93a22f29 geth[2908]: INFO [07-22|13:41:42.162] Imported new block headers               count=2    elapsed=15.593ms  number=12,876,513 hash=0c6ad5..f6ac58
Jul 22 13:41:45 ethereumonarm-f93a22f29 geth[2908]: INFO [07-22|13:41:45.178] Imported new block headers               count=1    elapsed=10.841ms  number=12,876,514 hash=8ba931..8a3b7b
Jul 22 13:41:50 ethereumonarm-f93a22f29 geth[2908]: INFO [07-22|13:41:50.210] State heal in progress                   accounts=735,632@37.24MiB slots=431,899@32.70MiB codes=662@5.17MiB nodes=10,122,472@3.34GiB   pending=26195
Jul 22 13:41:50 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:41:50.216] Unexpected trienode heal packet          peer=9f0aaa30 reqid=3,433,510,732,220,688,580
Jul 22 13:41:51 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:41:51.958] Unexpected trienode heal packet          peer=4d51d466 reqid=8,383,531,761,061,074,020
Jul 22 13:42:06 ethereumonarm-f93a22f29 geth[2908]: INFO [07-22|13:42:06.269] Imported new block headers               count=1    elapsed=14.490ms  number=12,876,515 hash=84f47f..5d1056
Jul 22 13:42:06 ethereumonarm-f93a22f29 geth[2908]: INFO [07-22|13:42:06.564] State heal in progress                   accounts=735,632@37.24MiB slots=431,899@32.70MiB codes=662@5.17MiB nodes=10,123,169@3.35GiB   pending=28510
Jul 22 13:42:06 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:42:06.565] Unexpected trienode heal packet          peer=c6330c95 reqid=7,889,053,466,489,058,058
Jul 22 13:42:06 ethereumonarm-f93a22f29 geth[2908]: INFO [07-22|13:42:06.990] Downloader queue stats                   receiptTasks=0   blockTasks=0   itemSize=195.18KiB throttle=1344
Jul 22 13:42:08 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:42:08.508] Unexpected trienode heal packet          peer=c161ec55 reqid=4,799,870,152,938,446,088
Jul 22 13:42:09 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:42:09.790] Unexpected trienode heal packet          peer=3221935c reqid=2,844,345,507,606,871,738
Jul 22 13:42:11 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:42:11.779] Unexpected trienode heal packet          peer=f9feef2b reqid=7,785,455,820,960,702,672
Jul 22 13:42:13 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:42:13.712] Unexpected trienode heal packet          peer=b3eaf04e reqid=5,356,255,396,390,595,891
Jul 22 13:42:14 ethereumonarm-f93a22f29 geth[2908]: INFO [07-22|13:42:14.677] State heal in progress                   accounts=735,632@37.24MiB slots=431,899@32.70MiB codes=662@5.17MiB nodes=10,123,532@3.35GiB   pending=29777
Jul 22 13:42:14 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:42:14.678] Unexpected trienode heal packet          peer=0ac5ad34 reqid=7,928,992,795,570,078,676
Jul 22 13:42:15 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:42:15.239] Unexpected trienode heal packet          peer=68aba13c reqid=7,030,748,983,030,171,115
Jul 22 13:42:16 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:42:16.109] Unexpected trienode heal packet          peer=e7fe0fd2 reqid=1,920,649,375,589,063,618
Jul 22 13:42:16 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:42:16.729] Unexpected trienode heal packet          peer=45f41019 reqid=7,548,212,257,530,585,929
Jul 22 13:42:17 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:42:17.365] Unexpected trienode heal packet          peer=55fd42f0 reqid=5,690,347,225,254,242,651
Jul 22 13:42:17 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:42:17.754] Unexpected trienode heal packet          peer=7bbccd97 reqid=7,890,019,142,461,179,849
Jul 22 13:42:18 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:42:18.211] Unexpected trienode heal packet          peer=8fe0dc37 reqid=666,678,050,563,697,745
Jul 22 13:42:18 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:42:18.335] Unexpected trienode heal packet          peer=61551118 reqid=1,545,041,451,227,207,801
Jul 22 13:42:18 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:42:18.479] Unexpected trienode heal packet          peer=b50ebd2e reqid=5,628,642,155,866,392,901
Jul 22 13:42:18 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:42:18.662] Unexpected trienode heal packet          peer=f296b062 reqid=7,088,508,473,366,170,741
Jul 22 13:42:18 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:42:18.745] Unexpected trienode heal packet          peer=c3c23622 reqid=626,377,224,210,211,409
Jul 22 13:42:18 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:42:18.905] Unexpected trienode heal packet          peer=5eaf53d9 reqid=4,995,603,858,963,162,143
Jul 22 13:42:20 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:42:20.826] Unexpected trienode heal packet          peer=9e25c006 reqid=1,191,199,376,667,411,249
Jul 22 13:42:21 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:42:21.078] Unexpected trienode heal packet          peer=251c5f30 reqid=4,537,057,820,273,556,430
Jul 22 13:42:21 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:42:21.133] Unexpected trienode heal packet          peer=d676db51 reqid=4,994,703,396,482,641,563
Jul 22 13:42:21 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:42:21.942] Unexpected trienode heal packet          peer=4edbfc5f reqid=7,119,699,682,441,132,004
[... cut ...]
Jul 22 13:42:35 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:42:35.137] Unexpected trienode heal packet          peer=4d51d466 reqid=8,383,903,165,527,450,593
Jul 22 13:42:35 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:42:35.159] Unexpected trienode heal packet          peer=c6330c95 reqid=2,226,942,420,973,443,774
Jul 22 13:42:35 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:42:35.217] Unexpected trienode heal packet          peer=c161ec55 reqid=7,834,552,617,822,423,573
Jul 22 13:42:35 ethereumonarm-f93a22f29 geth[2908]: WARN [07-22|13:42:35.287] Unexpected trienode heal packet          peer=3221935c reqid=1,496,881,092,591,068,813
```

diglos, 2021-07-23:
Hi. is it still ongoing?. I haven't any reference about this as we didn't run a snap sync yet (we will soon).

nimcon, 2021-07-23:
Still running. It's now been in heal for ca 3x as long as the initial sync took. What is the recommended way of syncing, if not via snap sync?

diglos, 2021-09-21:
Hi, any feedback?

nimcon, 2021-09-22:
Didn't manage to get the snap sync to work, and ended up doing a fast sync. Ran into some difficulties executing the rc.local script and had to run the commands from 'Ethereum software installation' onwards manually.

diglos, 2021-09-22:
Ok, thanks. If I may ask, why did you need to execute the rc.local script? Did the installation fail?

nimcon, 2021-09-22:
The installation didn't complete (afair there was no error message), but the changes to /etc/sysctl.conf had been made.

closed 2021-09-22 by diglos