EOA-Blockchain-Labs/ethereumonarm #15: Websocket Error

raahimSal, 2021-07-21:
When I try to connect to the geth websocket I am getting the following error:

2021/07/21 13:55:07 websocket: bad handshake (HTTP status 200 OK)

diglos, 2021-07-22:
Hi, 

Can you write here how do you try to connect to Geth?, are you using the systemd service to start it?

raahimSal, 2021-07-22:
yeah so the ws works when I use geth attach <ws address> on my pi. When I try to use my laptop to connect to the same WS it gives me the aforementioned error. The HTTP rpc works and I can connect to it on my laptop but just not the websocket rpc. I have tried both exposing the rpc to local network and open public and both give me the same error.

diglos, 2021-07-22:
Hi,

If you are accessing from a different machine try adding `--ws.addr 0.0.0.0` to /etc/ethereum/geth.conf and restarting geth

closed 2021-09-21 by diglos