EOA-Blockchain-Labs/ethereumonarm #23: Cannot access Pi through SSH

marinojoey, 2022-01-20:
I flashed my SD card and got about 33% through the sync when I got an error and I couldn't get it running again. No problem I said, Ill just re-flash and wait a few more hours.

I have now re-flashed via the same process and cannot login through ssh anymore. I have formatted the card 3 times and retried each time following the docs. The only deviation (that I made once) was adding an empty file named ssh in root for access.

Any advice?

Password being used: ethereum

Screen Shot 2022-01-17 at 11 29 33 AM

marinojoey, 2022-01-20:
[image]

marinojoey, 2022-01-21:
EDIT: I can access the pi via `ssh pi@111.111.1.11` and pswd: raspberry. 

I wrote the 'ethonarm_21.0.3.00.img' file onto the SD and included an ssh file in root. Otherwise I just plugged it in...

Also, logs are completely blank/static. Would appreciate any help!

diglos, 2022-01-21:
Ummh, seems like something went wrong with the flashing as the pi user is used with Raspberry OS and the Ubuntu image (which is our base OS) but not by our image.

Could you please mount the microSD and check if you are running Raspberry OS?

marinojoey, 2022-01-23:
Yea, I can definitely try again! I just want to be sure before I do tho. Below are my steps that got me to a point of logging in with raspberry OS credentials:
1) format and then write 'ethonarm_21.0.3.00.img' file onto the SD and included an ssh file in root.
2) format ssd and plug it in.
3) plug both into pi.
4) wait 15-20 minutes before attempting to login with `ssh ethereum@<ip-address>`?

I only ask because that is exactly what I am doing that is giving me ssh access through `ssh pi@<ip-adress>`. Thanks for the help!

marinojoey, 2022-01-25:
So i got around to trying it again and had the same results. I waited 30 minutes before attempting an SSH. Was unable to access it (per above) still. I was however still able to login through the default pi account. I used a new SSD, wrote the Eth-On-Arm .img file using the raspberry pi imager and added a ssh file in root. 

I have sha256 checksummed the file, so that is not the problem. I did not format the SSD this time, but I dont see how that would cause this problem.

Really confused by this one!

marinojoey, 2022-01-25:
Just formatted the SD again, wrote the img with Balena, ejected and plugged it in all per docs. Ssh connection was refused, so I waited an hour, pulled the card and added an ssh file. That allowed me to ssh through pi@<IPaddress> credentials again but NOT ethereum credentials. 

So I re-did everything once again, but this time I re-downloaded the zip from the docs. Checksummed it again, unzipped it and wrote to my SD. I added a root ssh file, ejected and plugged it into my pi. 

Waited 30 minutes and... it worked. I guess that this means that my img was somehow corrupted!

closed 2022-01-25 by marinojoey