EOA-Blockchain-Labs/ethereumonarm #26: Unsupported board type

Jeiwan, 2022-03-21:
I bought a new Raspberry PI 4 with 8Gb, and, apparently, it's not supported by the ethereumonarm image. The device won't boot: the green led keeps signaling 4 long and 4 short blinks, which means "Unsupported board type" ([source](https://www.raspberrypi.com/documentation/computers/configuration.html#led-warning-flash-codes)).

My `/proc/cpuinfo`:
```
Hardware	: BCM2711
Revision	: d03115
Serial		: 100000008bd5d2bb
Model		: Raspberry Pi 4 Model B Rev 1.5
```

Jeiwan, 2022-03-21:
It works on a newer version of Ubuntu: `ubuntu-20.04.4-preinstalled-server-arm64+raspi.img`

closed 2022-03-21 by diglos

gwenvador, 2022-04-02:
#28 I have the same issue but the upgraded version did not work.

gwenvador, 2022-04-02:
I downloaded from https://www.ethereumonarm.com/downloads/ethonarm_21.03.00.img.zip
Is this build updated?

gwenvador, 2022-04-02:
It worked with building its own image. so please update the download links in the docs with a newer image.

PR #27 (2022-03-21): Update Ubuntu version to support new RPI revisions
Fixes #26 

I'm running the new Ubuntu version on my Raspberry PI. Everything looks good:
```shell
ethereum@ethereumonarm-f9808bc20:~$ cat /proc/cpuinfo
...
Hardware	: BCM2835
Revision	: d03115
Serial		: 100000008bd5d2bb
Model		: Raspberry Pi 4 Model B Rev 1.5
ethereum@ethereumonarm-f9808bc20:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.4 LTS
Release:	20.04
Codename:	focal
ethereum@ethereumonarm-f9808bc20:~$ geth version
Geth
Version: 1.10.16-stable
Git Commit: 20356e57b119b4e70ce47665a719644 [... cut]