EOA-Blockchain-Labs/ethereumonarm #29: Invalid signature for http://apt.ethraspbian.com repo

0xalfalfa, 2022-04-14:
```
root:~# apt update
Hit:1 http://ports.ubuntu.com/ubuntu-ports focal InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease
Hit:5 http://apt.ethraspbian.com focal InRelease
Hit:6 https://packages.grafana.com/oss/deb stable InRelease
Err:5 http://apt.ethraspbian.com focal InRelease
  The following signatures were invalid: EXPKEYSIG 8A584409D327B0A5 Diego Losada Eth On ARM <diego.losadaf@gmail.com>
Reading package lists... Done
Building dependency tree
Reading state information... Done
24 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://apt.ethraspbian.com focal InRelease: The following signatures were invalid: EXPKEYSIG 8A584409D327B0A5 Diego Losada Eth On ARM <diego.losadaf@gmail.com>
W: Failed to fetch http://apt.ethraspbian.com/dists/focal/InRelease  The following signatures were invalid: EXPKEYSIG 8A584409D327B0A5 Diego Losada Eth On ARM <diego.losadaf@gmail.com>
W: Some index files failed to download. They have been ignored, or old ones used instead.
```

diglos, 2022-04-14:
Yes, this happens on previous installed images. Please run:

```
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 96B52AFDF8C1BD5E
sudo apt-get update
```

I'm leaving the issue open for now.

0xalfalfa, 2022-04-14:
Thanks a lot! That worked fine.

washosk, 2023-05-17:
focal repo is no longer supported, we're close it

closed 2023-09-14 by diglos