EOA-Blockchain-Labs/ethereumonarm #31: start4.elf is not compatible

HeroesDieYoung, 2022-04-28:
"start4.elf: is not compatible
This board requires newer software"

Stuck in a loop with this error on a brand new pi 4 model b. Is this not compatible with the latest board?

diglos, 2022-05-03:
Hi,

This happend with the old image. Did you download the newer one?

HeroesDieYoung, 2022-05-03:
Yeah okay, that has fixed the issue for me. I actually found the image from [this article](https://ethereum.org/en/developers/tutorials/run-node-raspberry-pi/), which looks like they've done a bit of setup to auto-start different services to separate an execution and a consensus image - but all based on an older image of yours. Starting from your page directly with the latest 22.04 based image is all good. That's a nicely written article and on ethereum.org I expect I'm not the only one to find the image that way, would be nice to get that article updated.

Either way, nice work.

closed 2025-07-04 by washosk