EOA-Blockchain-Labs/ethereumonarm #32: Fat Volume Not Supported

tbrent, 2022-05-02:
This is on the latest image `ethonarm_22.04.00.img`. I've tried re-flashing multiple cards and I get the same error. I see `0x0b` for the format type on the error screen, is that expected?

[image]

diglos, 2022-05-03:
Hi,

this seems a problem either with the Card or the card reader (or even with the Pi but this would be very odd). I would try flash it with a different utility:

https://www.balena.io/etcher/

Take a look at this as well:

https://forums.raspberrypi.com/viewtopic.php?t=294176

tbrent, 2022-05-15:
It's working now -- looks like flashing it with Etcher produced a different result than the command line. Maybe this had to do with using `conv=sync` instead of `conv=fdatasync`? I was flashing on a mac and didn't have `fdatasync` available. 

Thanks for your help! Closing this one out

closed 2022-05-15 by tbrent