EOA-Blockchain-Labs/ethereumonarm #36: Add support for more SBC

fgnm, 2022-08-06:
I was wondering to buy a SBC to run my own eth node, currently this project support Rasp Pi4, Odroid M1 and Rock5B.. But can't find the Rock5B anywhere, Pi4 is old and unjustifiably expensive, Odroid M1 not really good both on RAM and CPU.. Imo the Khadas VIM4 looks a good alternative for the Rock5B is it possible to support it too? Or even better, is there a dev guide to add support for other SBC?

Thanks!

diglos, 2022-08-06:
Hi.

We considered this board but we see 2 issues with it:

- It seems too expensive compared to Odroid M1 (we know is more powerfull than this one) and particularly with the Rock 5B (which we hope will be available in 2-3 months and even the 16 GB model is cheaper).
- The NVMe connection is x1 lane which makes absolutely no sense for a PCIe disk. Don't know why it was designed this way but it limits the disk I/O for this kind of disks.

If there is demand for it we may consider it again but I don't think is a good choice right now.

diglos, 2022-08-06:
That been said, you can easily adapt our script to the device if you have some Linux skills.

Take a look at the script here:

https://github.com/diglos/ethereumonarm/blob/main/image-creation-tool/ubuntu/Makefile

Basically it adds an /etc/rc.local bash script (which runs only at first boot) and add all config and software needed to turn the device into an Ethereum node.

fgnm, 2022-08-06:
I didn't check the NVMe connection... No sense at all.. So thanks for the explain... I'll try to find an Odroid M1 and later upgrade to the Rock5B.. Do you know how Odroid perform as a full node? Are performance comparable with a x86_64?

diglos, 2022-08-06:
Odroid M1 performs well if you take into account its cost. See this comparison table:

https://twitter.com/EthereumOnARM/status/1552596195796811776

fgnm, 2022-08-06:
Alright, thank you for all! Initial sync time it's not a problem as long as after that the node works well.. I'll start with an Odroid M1 hopefully to buy a Rock5B as soon as possible :)

closed 2022-08-06 by fgnm