EOA-Blockchain-Labs/ethereumonarm #38: ipfs not included with ethonarm_22.04.00.img.zip

lambo4jos, 2022-08-12:
The documentation here [Updating Clients](https://ethereum-on-arm-documentation.readthedocs.io/en/latest/user-guide/managing-clients.html?highlight=ipfs#updating-clients) says ipfs is an available package. I am running this image and noticed that ipfs was never installed, there is no ipfs service listed or running, and there is no ipfs.conf file included. Has this been removed or is it a bug and just missing from this particular img version?

diglos, 2022-08-12:
Hi.

Still supported but they changed the name of the package and need to adapt our scripts. Hope it will be available again in our repo soon.

diglos, 2022-08-21:
Hi.

It is now available. As of 0.14 version it was rename to Kubo:

https://github.com/ipfs/kubo/releases/tag/v0.14.0

Will update the doc and include it in the image ASAP but if you want to try it out:

```
sudo apt-get update && sudo apt-get install kubo
sudo systemctl start ipfs
```

You should forward the 4001 port.

Logs:

`journalctl -u ipfs -f`

lambo4jos, 2022-08-22:
Thanks for the update! I did already install ipfs and modified the service and config file and opened the ports, so I haven't been able to test this commit out myself on my current build. Looks like a great update though

By the way, great work on this project!

lambo4jos, 2022-08-26:
Just ran the update after removing my install of kubo, and everything seems to be working as expected. Great work

closed 2022-08-26 by lambo4jos