EOA-Blockchain-Labs/ethereumonarm #47: Instructions on how to install MEV

gwenvador, 2022-10-08:
Can you please give instructions on how to install MEV-boost on already installed raspberry. I can't find the package in the apt.ethraspbian.com repository.

thomasWos, 2022-10-08:
https://twitter.com/EthereumOnARM/status/1573338221224038402?s=20&t=iK6SmAqnTuHTFsqSR3pcnQ
Not sure the package is published...

You can have a look to this:
https://github.com/remyroy/ethstaker/blob/main/prepare-for-the-merge.md#choosing-and-configuring-an-mev-solution

And this to select a MEV relay
https://github.com/remyroy/ethstaker/blob/main/MEV-relay-list.md
(Maybe good to use a non-censoring mev-boost relay https://www.mevwatch.info/)

diglos, 2022-12-17:
Published last week:

https://ethereum-on-arm-documentation.readthedocs.io/en/latest/user-guide/mev-boost.html

closed 2022-12-17 by diglos