EOA-Blockchain-Labs/ethereumonarm #52: Rock5b sudo apt update doesn't support architecture armhf

lambo4jos, 2023-02-08:
When running `sudo apt update` on a fresh install on a rock5b using the latest image file ethonarm_rock5b_22.12.00.img.zip, there is an error message encountered that reads:

`N: Skipping acquire of configured file 'main/binary-armhf/Packages' as repository 'http://apt.ethraspbian.com focal InRelease' doesn't support architecture 'armhf'`

What I am curious to know is if this message is something that can be ignored or if this is something that needs to be investigated or fixed

3cl1ps, 2023-02-08:
got it too, ignoring.

lambo4jos, 2023-02-09:
> got it too, ignoring.

ignored it too, no issues so far

washosk, 2023-05-17:
it should be solved with the new apt repository.

closed 2025-07-04 by washosk