EOA-Blockchain-Labs/ethereumonarm #54: credentials : username: ethereum and password ethereum not working to login on image (O droid M1)

robertdavidwest, 2023-03-15:
Hello, 

I have been following the instructions to use this disk image in these docs: https://ethereum-on-arm-documentation.readthedocs.io/

Specially, I just used the disk image: ethonarm_odroid_22.07.00.img downloaded from https://ethereumonarm-my.sharepoint.com/:u:/p/dlosada/EejUgF6sH55EoUY3Pc34jwEBMIwIxYmJYDUqfGp0TJ1Eyw?download=1 to flash my SD card. When I then boot the device the credentials do not seem to work to login, not directly or via SSH. 

For comparison I just replaced the image with an Ubuntu server image from here: https://wiki.odroid.com/odroid-m1/os_images/ubuntu/20220318

and the login works fine both directly and via SSH. 

Sorry if this isn't the place to report this, I really wasn't sure where else to mention it. 

I'm flashing the drive on macOS Version 13.2.1 and installing on an O Droid M1 that I just purchased.

andro78, 2023-09-01:
I have the same issue, so I changed them in local shell.

binomialstew, 2023-10-01:
Sometimes, if the nvme is not accessible for whatever reason—like if there is not enough power to the device or if it wasn't flashed properly, it will not allow you to authenticate via ssh through the normal means because I believe it boots from the SD card with no home directory (because it is on the other missing volume). I have seen this when trying to power the pi through a usb port without enough power, and when it was changed to use the power supply, the nvme was mounted, and I was able to authenticate as usual.

closed 2025-07-04 by washosk