EOA-Blockchain-Labs/ethereumonarm #58: [bug] Dashboards not working

emlazzarin, 2023-05-12:
A fix that also worked for me, from r3dDoX in the Discord:

> After checking the prometheus.yml I saw that the configs should all be under: /etc/prometheus/files_sd. This folder is not there on my machine though. So I copied it from the github repo under: fpm-package-builder/utils/ethereumonarm-monitoring-extras/sources/usr/lib/ethereumonarm.  And now my Grafana Dashboard seems to slowly get data in 👍 

I can look into fixing this at some point, but wanted to put it here in case someone else wants to.

emlazzarin, 2023-05-12:
Wondering if the issue is here: https://github.com/diglos/ethereumonarm/blob/b32cc5687137ba0f6851fe2cfdbc266a978cbf0a/fpm-package-builder/utils/ethereumonarm-monitoring-extras/sources/usr/local/sbin/set-ethereunonarm-monitoring-extras#L13

r3dDoX, 2023-05-13:
Additionally to the fix you mentioned above, I had two more problems on my dashboards:

1. On the dashboard `04-EoA Node Status` I had a lot of errors saying the duration format was wrong. I saw the use of the variable: `$__range_interval` in the queries, but I think the interval variable should be: `$__interval`. So I replaced this variable on the whole dashboard in the JSON model. This seems to work fine now.

2. Some queries are using the function `last_over_time` which is not recognized. I researched a bit and saw that this was introduced with prometheus 2.26. But the installed prometheus version seems to be: 2.15.2. I wasn't yet able to fix this one.

r3dDoX, 2023-05-15:
@emlazzarin I might have time towards the end of this week. Haven't really read into this repo yet. But I could go from your pointer above and check if I can find out what's happening

washosk, 2023-05-17:
> Wondering if the issue is here:
> 
> https://github.com/diglos/ethereumonarm/blob/b32cc5687137ba0f6851fe2cfdbc266a978cbf0a/fpm-package-builder/utils/ethereumonarm-monitoring-extras/sources/usr/local/sbin/set-ethereunonarm-monitoring-extras#L13

Yes it was the problem, I just upload a new version fixed. Thanks for your help

washosk, 2023-06-11:
In the next days we're gonna upgrade Prometheus version  to the last stable one, and add new dependencies to dashboards, it should fix everything

closed 2023-09-02 by diglos