EOA-Blockchain-Labs/ethereumonarm #59: Is this repo maintained? tests on rpi don't work

MicZet, 2023-06-22:
hello!
is this repo maintaned? 
i followed instructions for installation but seems it doesn't work
even clients are not being installed and of course services (like lighthouse) cannot be started...

in logs i see the eth repo was installed properly but then installlation of packages fails, for example:
Jun 22 05:33:18 ubuntu rc.local[2243]: Package eth2.0-deposit-cli is not available, but is referred to by another package.

so not really usable :(

diglos, 2023-06-22:
Hi.

Yes, the Rpi image needs an update as well as the other devices. We are working on it for all boards but we are seriously considering dropping support for the Rpi 4 as it is really hard to run a node on this device nowadays. Unfortunately, the hardware requirements raised after the merge  and this put the Rpi in a very hard position.

Besides, there are other devices that have similar cost (even lower) and are much more powerful such as the Rock 5B and the Orange Pi 5 plus.

We will most probably give the Rpi image a last update but take into account that it is not the best option to run an Ethereum node.

lokoo, 2023-07-05:
Damn... I just bought a raspberry pi 4 for an ethereum node. Wish this was known and ethereum.org wouldn't mention it. No clue whats going wrong, but Im getting: `Failed to start lighthouse-beacon.service: Unit lighthouse-beacon.service not found.  `
Seems like setup just doesn't work.
Well too bad. I'll try to set it up myself then from a clean Ubuntu I guess.

cynic-1, 2023-07-23:
Yeah. I got the same problem. Setup just didn't work.
I'm trying to set it up. If you get it right, maybe you could share with me?
Thanks a lot.

> Damn... I just bought a raspberry pi 4 for an ethereum node. Wish this was known and ethereum.org wouldn't mention it. No clue whats going wrong, but Im getting: `Failed to start [...]

closed 2023-09-02 by diglos