EOA-Blockchain-Labs/ethereumonarm #65: Validator graphs not showing data in grafana

binomialstew, 2023-10-01:
I have a running validator, and most of my other data is showing in graphs through grafana, but for some reason, my validator graphs are missing data. Am I missing something? I can't seem to find any specifics around this in the guide.
[image]

diglos, 2023-10-02:
Hi,

Yes, dashboards are currently broken, we are trying to work out a solution for all clients, sorry about that.

In the meanwhile, try to find a dashboard for your consensus client and import it directly to Grafana.

emkman, 2024-01-02:
How do we start Grafana? I just did a fresh install and there is nothing on port 3000. Do I have to manually install `ethereumonarm-monitoring-extras`?